RaspberryPints/RaspberryPints #302: Should not use mysql functions

jksnetwork, 2015-03-31:
The old mysql functions are deprecated and shouldn't be used anymore.  Should be replaced with mysqli or pdo_mysql

jksnetwork, 2015-03-31:
If I wanted to fix this issue which branch should I work on?  v2?

jksnetwork, 2015-03-31:
Also is there any reason for having 3 separate files with db credentials?  includes/config.php, admin/includes/configp.php and admin/includes/conn.php

mcangeli, 2015-03-31:
There were several reasons for using the old mysql functions. I believe
first and foremost was the fact that the base code RP is based on was using
them. We kind of just stuck with it at that point. In addition, it allows
backwards compatibility with systems running older versions of mysql.

jksnetwork, 2015-03-31:
I am going to clean it up a bit.  The weird thing is that the install scripts were using mysqli, the personalize page was using PDO and the rest all normal mysql functions.  I have started implementing a mysqli db class that uses prepared statements and re-writing the queries.  I got through the front end and most of the admin pages last night.  I just need to hit admin/includes and the install scripts.

mcangeli, 2015-03-31:
Yea, the Install script was coded after the original site. The developer
used the mysqli code.

jksnetwork, 2015-03-31:
Should i branch off the v2 branch or the develop branch?  Having a  hard time figuring out the workflow here.

mcangeli, 2015-03-31:
Version2.

The development branch has none of the flowmeter code in it.

jksnetwork, 2015-03-31:
Good since thats what I did the work in already :)

Skemp45, 2015-03-31:
Yeah, we orginally all took parts and built them out with different things in mind and combined at the end of the road and just released it. Redoing everything in classes was on the roadmap. Nice work on the new updates, one of us will review it all and if its all working, we will pull it in to the next release!

closed 2016-01-15 by Skemp45