RaspberryPints/RaspberryPints #304: Revamp time! Whats Next?

Skemp45, 2015-03-31:
We have all taken about a year off of this project and have not made much movement on it. Who all is still around? 

What should our next steps be?

thadius856, 2015-03-31:
I'm still around.

Have had some big life changes in the past year so time has been scarce for
development. Never did get around to getting my second alamode.

Since then the new RPi has come out and likely many Raspbian builds.
Anybody know if we're still compatible as-is?

I just finished building a new BSD server, so I'm a lot more confident in
my *nix skills now. I haven't done much with PHP, CSS, or SQL since.

I think a revamp to a new stable build is probably in order by now. And our
V3 goals should probably be re-evaluated based on user feedback and
conprtition, with a few intermediate versions before we add those features.

I'll make as much time as possible for this project, but thinking we need a
face-of-the-company type person brought into the mix for frontline support
to really get rolling again.

mcangeli, 2015-04-01:
Who the hell woke Skemp up??

:)

I'm here. Since the Flowmeter release in September, I'm on my second job
already....

I have the basis of an RFID tracking system working to count pours to users
(its a thing of beauty) but I don't have a back end to add the rfid to the
user. Its currently a convoluted process.

I think we do need to sit down and look at a roadmap.

V3 *should* look to cleaning up the code a bit and fixing the current
issues it has before we add more features.

Boombrewer, 2015-04-01:
I'm still lurking in the shadows but not sure how much time I'll have to help out.  Just bought a house, moving, and I have a 5 month old.  The only reason I still get brew days is that the wife loves keeping the kegorator full.... hours on the computer at night, not so much.

I'll do what I can to help out though.

jksnetwork, 2015-04-02:
I'm new to the project but I could help out as well.

Skemp45, 2015-04-02:
I agree we need to be able to support and troubleshoot users issues but the forum was just overwhelming, along with alot of issues that were not code based issues. We need someone or someone(s) to do all the troubleshooting on the form and escalate true bug issues up to the dev team (us) and we can work them how we see fit. Are any of you familiar with the Agile methodology?

jksnetwork, 2015-04-02:
I am somewhat familiar with Agile. The devs at work use it and I've sat in a a few conference sessions on it.

Skemp45, 2015-04-02:
Nice, we need to really get organized and but stuff out in sprints instead of taking on everythign at once. 

On another note, EVERYONE NEEDS to read this writeup along with comments I found:
http://www.reddit.com/r/Homebrewing/comments/2haipu/review_kegberry_vs_raspberrypints_wo_flowmeters/

it points out alot of areas we are lacking and what we do well, this should help us move forward

jkurl, 2015-04-02:
Sean,   I called this need out a long time ago about providing support and let you devs work w/ out being overwhelmed by coding and troubleshooting other issues w/ the system. I am not trying to be a dick about it in any way. I am back from deployment and on vacation until May. I like boombrewer can offer help but it is limited. I am retiring the end of this month and working on transitioning back to the civilian life.Jerry Sent using CloudMagicOn Wed, Apr 1, 2015 at 19:48,  RaspberryPints/RaspberryPints <reply@reply.github.com> wrote:            Nice, we need to really get organized and but stuff out in sprints instead of taking on everythign at once. 

On another note, EVERYONE NEEDS to read this writeup along with comments I found:
http://www.reddit.com/r/Homebrewing/comments/2haipu/review_kegberry_vs_raspberrypints_wo_flowmeters/

it points out alot of areas we are lacking and what we do well, this should help us move forward

Skemp45, 2015-04-02:
@jkurl I remember you saying exactly that, the problem is like you me and most everyone is dedicated support is hard when its all just for fun in the free time, but we need to figure something out and I feel that together we can! IDEAS?

Skemp45, 2015-04-02:
I know @day_tripper on HBT has been our saving grace on the forum since day one, he knows a lot about Rpints now and is constantly answering everyone's questions

jkurl, 2015-04-02:
One idea I had was maybe do a 2 tier type system of support. The tier 2 support will be easier issues like hardware and some lower level software support. Then tier 1 would be someone like day_tripper who can go a little deeper in the software side and the hardware side as well. Sent using CloudMagicOn Wed, Apr 1, 2015 at 20:26,  RaspberryPints/RaspberryPints <reply@reply.github.com> wrote:            I know @day_tripper on HBT has been our saving grace on the forum since day one, he knows a lot about Rpints now and is constantly answering everyone's questions

mcangeli, 2015-04-02:
I looked at kegberry (I have their install script that I was working on
basing one for rpints off of).

Its comparing apples to grapples.

Yes, they do the same thing but in different ways.

If I remember, kegberry is written almost completely in python. Its based
heavily on kegbot and is no where near as customizable as rpints. Thats
where we stand out.

Don't like the layout? Change it. We'll even help.
Want to see a new feature? Develop it on your own (motion sensor) or let us
know and we'll work on it.

Kegberry seems to be more for the person that has a kegerator at home that
they fill with stuff from the store.

Either way. I don't see it as a competition. We aren't selling a product.
We're a bunch of DIY'ers who are working on a project because we want to.
We don't want to use something someone else develops and maintains.

I like Rpints because if I want to change the way something works, I can
change it. If I want to add a feature, I can.

But thats just me.

Skemp45, 2015-04-02:
I agree with a lot of your points, but there is no reason we cant make changes and new releases and be the best product for both apples and grapples. We are not selling a product and we are DIY'ers but that does not mean our project should lack

olywrestle, 2015-06-25:
Totally new guy here, don't know a ton about coding but am learning. Already running pints and have made some simple tweaks. Have not gotten all the parts for flow meters yet, but will.

Just wanted to jump in here and let you know the new BJCP styles are out, and I was going to modify/update the styles database with the new ones unless someone already has. 

Let me know if there is anything I can do to help out/test.

Skemp45, 2015-06-25:
Do you have a link? I can roll that into the next release

olywrestle, 2015-06-25:
http://bjcp.org/docs/2015_Guidelines_Beer.pdf

closed 2016-06-13 by Skemp45