RaspberryPints/RaspberryPints #306: Read This! V3 Breakdown

Skemp45, 2015-04-02:
RPints V3 Release breakdown
Main Focus:
1.	Cleaning up code 
2.	Making Rpints more user friendly
3.	Make install 1000 times easier
4.	Inventory section in admin panel
5.	Analytics/tracking users want this
6.	Add function to allow images with beers, such as labels
7.	Fix issue with HAVING to have og fg and allow ABV
How will we do this:

We need to sit town and break this out into tasks then into sprints where we decide in a 2 week period what tasks we will be able to get done and push to finish those tasks. Our users are loyal and want a “better” product and we can give that to them. 
•	We need to really clean our code to increase performance, security, and to stay with the times. This will also allow us to develop easier in the future. 
•	We also need to focus on how things look and flow, currently things are choppy and not very intuitive, LETS CHANGE THAT. 
•	Installing Rpints is a PAIN IN THE ASS for non unix users, lets make a script to handle the whole LAMP and other dependencies, then use the Rpints installer to install rpints so the user only has to fill out maybe 2-3 forms instead of running 20+ commands. 
•	We can easily add a section where they can add beers they have in kegs waiting to be tapped and give an option when tapping a beer to choose from inventory.  
•	We already track the amount left in a keg we need to find other ways to track information and report this back to the user. 
•	People like images so lets give them the option in an elegant manner, allow images to go with beers
•	Not all kegs might be home brew lets allow for commercial brews too!

LETS DO THIS!

Skemp45, 2015-04-02:
is there any disagreements on the direction I think we need to move for V3?

mcangeli, 2015-04-02:
> RPints V3 Release breakdown
> Main Focus:
> 1. Cleaning up code
> 2. Making Rpints more user friendly
> 3. Make install 1000 times easier
> 4. Inventory section in admin panel
> 5. Analytics/tracking users want this
*Can't do this without a way to track/add users. *

> 6. Add function to allow images with beers, such as labels
> 7. Fix issue with HAVING to have og fg and allow ABV
> How will we do this:
>
> We need to sit town and break this out into tasks then into [...]
*** I've begun the work on this..*.

> • We can easily add a section where they can add beers they have in kegs
> waiting to be tapped and give an option when tapping a beer to choose from
> inventory.
>
> • We already track the amount left in a keg we need to find other ways to
> track information and report this back to the user.
***Need to decide **what** information you want to report back to the
user. *

> • People like images so lets give them the option in an elegant manner,
> allow images to go with beers
***I have this working providing the "homebrewery" is setup on untappd. It
could be done locally (allowing a user to upload an image/logo for a beer) *

> • Not all kegs might be home brew lets allow for commercial brews too!
*There isn't really anything stopping someone from having a *non* homebrew
beer as it currently stands, the change from OG/FG to ABV would help. *

> LETS DO THIS!
***Lets not and say we did?***

closed 2016-06-13 by Skemp45