RaspberryPints/RaspberryPints #311: Anyone Still Around?

Skemp45, 2016-01-15:
It is been a bit and RaspberryPints has been pretty much running itself on auto pilot. How is everyone doing?

mcangeli, 2016-01-15:
Been busy. Haven't had time to work any more on the RFID stuff (but its
working...)

Skemp45, 2016-01-15:
Yeah, I have been getting about 1 support question a week so I know people are still using it haha. I have not decided if I want to undertake a remake or not on this.

thadius856, 2016-01-15:
I'm out. Going through some pretty major life changes and don't even have
time to brew.

I've made huge leaps on my sql, python, and vba skills through work, but
I'm not at a point right now where I can spare much time.

Let me know if you guys want the GitHub account handed over to someone.

Skemp45, 2016-01-15:
@thadius856 I will take it over, I still control the web hosting and domain etc. shawn@besmartdesigns.com just email the details on over

joahfitzgerald, 2016-01-15:
I'm pleased to see this still has some collaboration.   Thanks to all who
have participated thus far.

Boombrewer, 2016-01-18:
Sorry guys, I'm out.  Too many moves by Uncle Sam and the system does not prove robust against movers.  I'm sad to say I'll be disassembling when I can get some spare time and have a few flow meters and such for sale.  Good luck to you and hope to rejoin when we get a little more stable.

Skemp45, 2016-01-19:
No worries to everyone who can no longer assist, It has been two years and a lot of changes have come through. What we were able to do as a group of strangers is nothing short of amazing. I plan on taking Rpints and re writing most of the code to current standards and design a whole new layout along with a very very very easy install script. I will most likely start selling pre configured pi's with the new version. I will keep yall updated with any changes along the way.

Best Regards,
Shawn

thadius856, 2016-01-19:
Just let me know if you do end up selling so I can remove the link from my
HBT account. They said they'd ban me if I was linked to our page and
started selling or collecting donations without a vendor account.

RaspberryPints, 2016-01-19:
I will for sure, if I go that route I will probably get the proper merchant account too.

thadius856, 2016-01-30:
@Skemp45 Can you remove me from the help email distro for now?

Skemp45, 2016-01-30:
Yes of course I will

Scimmia22, 2016-08-04:
So is this dead now? Things closed with no resolution, PRs sitting forever...

jcorrado, 2017-07-24:
Hello,

This project seems to have wound down, but if any of the original devs are around, I just wanted to say thank you as I used your Arduino code in my own, toy project.  Attributed, of course.

Thanks,
Jereme

Philusha1983, 2017-07-24:
Good to see people are interested.
I'll be happy to see what you made from this code.
I'm following this project for awhile looking for some spare time to
implement it in my kegarator.

Skemp45, 2017-07-24:
HEY! I am still here just unable to dedicate time to it at the moment. I am very happy you were able to use some of our code! Love to see people making stuff!

jcorrado, 2017-07-25:
Thanks, @Skemp45  - you guys saved me a ton of time!

@Philusha1983 I just recently began learning Clojure and needed a little project to tackle as an exercise... something that would benefit from a multi-threaded solution.  At around that same time, someone in my office asked me if the coldbrew keg could tweet in the morning, so people knew if they should pick up their own iced coffees or not :)

I build a little HTTP API to surface basic keg stats: [know-the-flow](https://github.com/jcorrado/know-the-flow).