RaspberryPints/RaspberryPints #313: Please Help

drew7544, 2016-03-28:
Im at my wits end.
Cant seem to get past this pint in Raspberry Pints installation

Step6.
sudo mv RaspberryPints-x.x.x.x/* /var/www/

Moving the files to web host folder. Im putting in the version number 2.0.1, but getting this message
pi@raspberrypi:~ $ sudo mv RaspberryPints-2.0.1/* /var/www/
mv: cannot move ‘RaspberryPints-2.0.1/admin’ to ‘/var/www/admin’: Directory not empty
mv: cannot move ‘RaspberryPints-2.0.1/arduino’ to ‘/var/www/arduino’: Directory not empty
mv: cannot move ‘RaspberryPints-2.0.1/img’ to ‘/var/www/img’: Directory not empty
mv: cannot move ‘RaspberryPints-2.0.1/includes’ to ‘/var/www/includes’: Directory not empty
mv: cannot move ‘RaspberryPints-2.0.1/install’ to ‘/var/www/install’: Directory not empty
mv: cannot move ‘RaspberryPints-2.0.1/python’ to ‘/var/www/python’: Directory not empty
mv: cannot move ‘RaspberryPints-2.0.1/sql’ to ‘/var/www/sql’: Directory not empty

Nothing is working after this step.

Please Please help me

mcangeli, 2016-03-28:
try sudo mv -f RaspberryPints-x.x.x.x/* /var/www/

drew7544, 2016-03-30:
@mcangeli 
thanks but i tried that and got the same error as above.

i checked in file manager and the html file is empty?

Skemp45, 2016-06-13:
@drew7544  I would suggest re importing that zip and trying again, something may have gone corrupt in the process.

closed 2016-06-13 by Skemp45