RaspberryPints/RaspberryPints #314: Flow Meter Setup Help - Step 6

aaronorwick, 2016-04-06:
First of all - Raspberry Pints is freaking sweet!  A relative beginner such as myself was able to get most of this up and running.  The only part that I haven't been able to get going is the alamode setup to add on flow meters.  For some reason I get an error when trying to run the setup - what am I doing wrong?

pi@raspberrypi ~ $ tar -xvzf alamode-setup.tar.gz
alamode-setup/
alamode-setup/80-alamode.rules
alamode-setup/arduino-setup
alamode-setup/avrdude
alamode-setup/avrdude.conf
alamode-setup/boards.txt
alamode-setup/inittab
alamode-setup/jessie-setup
alamode-setup/setup
pi@raspberrypi ~ $ cd alamode-setup
pi@raspberrypi ~/alamode-setup $ sudo ./setup
sudo: unable to execute ./setup: No such file or directory
pi@raspberrypi ~/alamode-setup $ ls
80-alamode.rules  avrdude       boards.txt  jessie-setup
arduino-setup     avrdude.conf  inittab     setup

Thanks for any help that can be offered - cheers!

Huhwha, 2016-04-10:
I'm stuck at exactly the same point. This is all new to me, so I definitely need help!

aaronorwick, 2016-04-25:
Hi - is there anyone out there that has run into this and solved?  Thanks!

Huhwha, 2016-04-25:
Sorry - I meant to post back here.  Bottom line is that alamode-setup file is hosed.  Replace it with the one here and you should be good to go.  

"Been getting lots of help the last couple of days via PM... thought I'd return the favor. 

https://drive.google.com/file/d/0Bzp...ew?usp=sharing

 Uploading Wheezy, the good alamode sketch, and anything else folks think would be helpful..."

wiseosagie, 2016-05-09:
I'm having the same issue. can anyone assist?

wiseosagie, 2016-05-09:
Huhwha, the link doesn't work

closed 2016-06-13 by Skemp45