RaspberryPints/RaspberryPints #323: Anyone still out there?

fishbum, 2019-01-15:
Hi all, I know this is really old, I'm having issues installing because php5 seems to be obsolete.  I installed php7 (or 7.0) but it's not working.  I get this probably isn't supported but if someone would be kind enough to just point me in the right direction, I'd be grateful.

Skemp45, 2019-01-15:
Hello! Yeah we have not been able to update the software since the last release on October of 2014 due to time constraints. I did start building a new paid version but i'm not sure there is enough interest in a paid version to push through and finish it.  

We do still have an active thread going on homebrew talk where the community seems to be helping each other get through many issues with the dated code. Here is a link to it:
https://www.homebrewtalk.com/forum/threads/version-2-release-raspberrypints-digital-taplist-solution.487694/page-102

I really hope this helps!

fishbum, 2019-01-15:
Thank you so much for the reply, I’ll check the thread out!!

closed 2019-01-15 by fishbum